jblanked/Picoware #28: Features Requests

guidouil, 2025-09-01:
#FR1 SSH Client

cvwong, 2025-09-13:
Can add boot wallpaper customization?

theking8713, 2025-11-12:
> #FR1 SSH Client

This would be my ultimate goal for this device, either with the pico2w or future revisions or with some other pico compatible MCU or maybe tiny SBC, i have in inventory a pico, a pico2w and a luckfox Lyra for this purpose, 
Iv also been trying to make this "modern dumb terminal" concept a reality for years now, iv got kits from pimironi for the pico that instantiate vga and a usb host but those don't leave enough gpio for adding wifi, I haven't rebuilt any of the firmware for the newer wifi enabled pico and this effort was a while ago,  before the pico 2 or 2w were ever released and the ultimate limitations ended up being resolution related and the complexity of initiating SSL for secure shell on such a limited MCU..
Iv also got several ttgo vga32 v1.4 boards as I was hoping that those would end up being an out of the box solutions to this pipe dream but again just far to limited resource wise to make this particular eccentric idea come to life...
At the time I was hoping to put together a minimul desktop style "dumb terminal" on modern hardware like a big flat panel and mechanical keyboard, do need a mouse to work from a CLIs of my many many servers and I wasn't even needing complex colors just maybe 2 or 3 basic ones to customize some context highlighting in the interface otherwise as far I was concerned it could be monochrome so that should have made the memory needs of a larger resolution frame buffer easier to pull off, but alas at the time I just resolved to using a pi zero 2 w and embedded that into a thick 23in monitor with access to IO for a mouse and I also added 100M ethernet because I thought that was prudent and kinda gave up on the idea of a MCU based dumb terminal...
Untile I came across clockworkpi device videos on youtube and that lite the spark again, lol
I unfortunately did pay the extortionate price for the let down that is the uconsole that NEEDs a revision to the keyboard like yesterday, and better out of the box wifi antenna options... So that is sitting in a box with the cm3+ and the cm4 and it's io board untile something changes on that front, 
But the picocalc... There is hope for this thing yet ..
I have no doubt that the requirements of a ssh client will far surpass the capabilities of these pico class MCUs but I'm also not above hosting a low resource telnet server that act as a telnet access point, possible on one of these many wifi enabled pi zeros or equivalent class devices i have from orange pi and radxa and nanopi
But I'm in the wait and see path for what we might be able to put together on this platform...
I like the feel of it, it's size, the screen, the keyboard and the overall early 2000s graphics calculator feel... It's nice, let's keep building on this one ya?